hoantv/CarSimulatorFirmware #6: PWM FFB Review

manoukianv, 2020-06-22:
Hello,

i made a PWM Review. 

I think you have to use a DAC because in the main file, you don't initialise PIN with the TIM fonction like that : GPIO_PinAFConfig(GPIOE, GPIO_PinSource9, GPIO_AF_TIM1) in the main.c program around line 226. So without this pin enabling function, PWM signal is not generated, and i think, you need to declare the DAC to force the pin.
You would remove adc, it will not help at all.

in your "run" pwm method :  I think you have to setup first the DIR before the PULSE pin. As you have to do job twice (setup the PWM and the DAC), i think you have several clock that allow PWM to generate a quick not wanted signal.

=> In summary, what you could do : 
1. Remove the DAC generation and use only PWM generator
2. Setup DIR before clock.

Vincent

manoukianv, 2020-06-22:
I check the pinout setting, and i confirm the issue in the documentation.
PE9 : CH1 - Pulse
PE10 : CH2N - Dir comp.
PE11 : CH2 - DIR

When i use this settings (PE9/ PE11) the only configuration to make works the motor is tu used "reverse encoder" and controller direction in "reverse", else on calibrate motor not start to pin, or spin infinitly, i check the hardware and there is no issue. I"m sure it's because you generate dir in the opposite direction ;) On CH2N it's work with the good setting but it"s the opposite pin ;)

So you have too solutions : 
  -> Reverse the 0/1 logic, change the doc    👎 not compliant with MMOS pinout
  -> Stay like that : don't reverse logic and the doc 👍 compliant mmos, even if using CH2N, we don't care, it works :-p

manoukianv, 2020-06-22:
I generated a sample for TIM1 in PWM, and i confirm you, that to works, it needs to initialise the PIN. This is done in the st32f4xx_hal_msp.c with this code  : 
`void HAL_TIM_MspPostInit(TIM_HandleTypeDef* htim)
{
  GPIO_InitTypeDef GPIO_InitStruct = {0};
  if(htim->Instance==TIM1)
  {
  /* USER CODE BEGIN TIM1_MspPostInit 0 */

  /* USER CODE END TIM1_MspPostInit 0 */
  
    __HAL_RCC_GPIOE_CLK_ENABLE();
    /**TIM1 GPIO Configuration    
    PE9     ------> TIM1_CH1
    PE10     ------> TIM1_CH2N
    PE11     ------> TIM1_CH2 
    */
    GPIO_InitStruct.Pin = GPIO_PIN_9|GPIO_PIN_10|GPIO_PIN_11;
    GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
    GPIO_InitStruct.Pull = GPIO_NOPULL;
    GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_LOW;
    GPIO_InitStruct.Alternate = GPIO_AF1_TIM1;
    HAL_GPIO_Init(GPIOE, &GPIO_InitStruct);

  /* USER CODE BEGIN TIM1_MspPostInit 1 */

  /* USER CODE END TIM1_MspPostInit 1 */
  }

}`
If you don't have this, add it like i explain it in the first message, and remove the DAC....

hoantv, 2020-06-24:
- Pwm was solved in V0.17
- DAC keep to support DD for testing long time 1st.

closed 2020-06-24 by hoantv