hoantv/CarSimulatorFirmware #12: source code

Peppazzz91, 2020-07-09:
is possible to have source code for porting on stm32 blue pill (STM32F103C8T6 based)?
It was capable to run a ffb firmware (i know a **closed** source firmware thath works, https://www.facebook.com/EMCDeveloper/), but pin used in MMOS does not exist on this board

hoantv, 2020-07-09:
this firmware can not be used in STM32F103C8T6 :).

Peppazzz91, 2020-07-09:
for what reason?

hoantv, 2020-07-09:
They have different hardware architecture :)!

Peppazzz91, 2020-07-09:
sorry, i made a mistake.
I read this
[image]
so i think that is an open firmware with source code and that works on stm32 cortex "m" based architecture...........................

closed 2020-07-09 by Peppazzz91