hoantv/CarSimulatorFirmware #27: "Unable to save data" when attempting to change Motor Controller type

HighMasterGogo, 2020-12-27:
Hardware tab -> Motor section -> Controller drop down -> Select "Internal DAC"

Click Apply

Click Save

Debug reports...

`[16:25:31] [Motor]Data successfully sent (42) bytes
[16:25:31] [Save-11]Data successfully sent (9) bytes
[16:25:36] Unable to save data`

Go back to drop down and it remains on "PWM".

chicknsoup, 2020-12-28:
Thank you for your report. To be able to detect the problem, please spare some time to fill out the below form:
<pre>
1. Brief description of the issue
2. Which firmware and UI version are you using?
3. Windows version?
4. Which feature is broken?
5. Does it occur frequently? (Rarely, sometimes, always, etc.)
6. Steps to re-produce the issue?
7. Screenshots and logs
8. Other information
</pre>

HighMasterGogo, 2020-12-28:
This was an error with versioning names. The issue was with V.0.21.1.4 (which I believe is a typo).

V.0.20.4.0 is the correct version and does not have this issue.

closed 2020-12-28 by HighMasterGogo