hoantv/CarSimulatorFirmware #30: e-Stop Leds

Mageign, 2021-01-26:
Maybe add support for external led (not the ones on the board).
One for e-Stop engaged, another for e-Stop disengaged ?

hoantv, 2021-01-27:
You mean stm32 discovery? Next version > 2.3 i added 2 led for some information. But with estop, have to becareful with it :)!

Mageign, 2021-01-27:
Yes Stm32 discovery.
What is the problem with e-Stop ?
I just think a led to indicate e-Stop is engaged, and another for disengaged would be great.

hoantv, 2021-02-01:
With e-stop, you must connect with external power. So it doesnt relate to stm32. Connect led directly with estop. 
this feature does not need on stm32.

Mageign, 2021-02-01:
How can you connect 2 leds with 2 wires ?

hoantv, 2021-02-02:
> How can you connect 2 leds with 2 wires ?

Connect a led continuously with estop button

Mageign, 2021-02-02:
So it is not possible to connect 2 leds ...

groybe, 2021-02-07:
> So it is not possible to connect 2 leds ...

It is possible. Depend what you have wire estop to. Connect one led to normally closed and one to normally open terminal on estop.

hoantv, 2021-02-08:
This doesnt relate to firmware. So i close it.

closed 2021-02-08 by hoantv

Mageign, 2021-02-08:
I hoped you could add it in the firmware.
Thanks for your work ...