hoantv/CarSimulatorFirmware #45: Pedal calibration and preset save problem!

EMrtr22, 2024-06-23:
Pedal calibration and preset does not save settings.

It is necessary to calibrate again when the USB connection is disconnected.

Ui : v4.4.2.2 & Frimware : 1.0.5.1

chicknsoup, 2024-06-23:
Please provide the UI logs folder. Thanks.

EMrtr22, 2024-06-23:
> Please provide the UI logs folder. Thanks.

[application.log](https://github.com/user-attachments/files/15943909/application.log)

chicknsoup, 2024-06-23:
> > Please provide the UI logs folder. Thanks.
> 
> [application.log](https://github.com/user-attachments/files/15943909/application.log)

This might be firmware related. @hoantv

EMrtr22, 2024-06-23:
> This might be firmware related. @hoantv

Thank you for reply, i hope the problem is solved.